alexforencich/xboot #16: ATmega32u4

maziarzamani, 2015-01-21:
Is it possible to use xboot on ATmega32u4?

alexforencich, 2015-01-21:
It should, but xboot does not support USB booting so you'll have to use one of the serial ports.  It's possible the makefiles are out of date and need some extra information for the 32u4.  There are other bootloaders that only support USB booting on xmega, though, so if you wan to boot via the onboard USB controller then another bootloader might be a better idea.  However, I think xboot is the only one with an API.  Not sure if you need that feature or not, though.

maziarzamani, 2015-03-23:
I was planning to use the existing I2C bus which i already use for data transmission for flashing the program. Would that be possible? And how do i flash xboot along with `Caterina-Micro.hex` (Arduino bootloader for Arduino Micro)