alexforencich/xboot #19: API not responding

MADHUKARAVS, 2015-09-25:
Hi there,
I gone through the link and downloaded the code
Created Xboot.h hex file( enable UART and A LED for communication, Enable the API)
Programmed the Xboot.hex file and i can see the LED blinking and also UART responds with Xboot++ when i send 'S' through UART.

My problem is whenever i call Any API (get version) from my application code, its restarting
am i doing anything wrong?

alexforencich, 2015-09-25:
No idea.  Are you building your code with the xboot makefile?  Have you read out the chip after writing the bootloader to make sure it was written to the proper location?  Can you print out the jump table offset and confirm that the jump table is sitting at that location after programming?

MADHUKARAVS, 2015-09-25:
Thank you for your quick response
After Programming the Xboot.hex, i checked if its sits in the proper location.
For me It looks like its situated in proper location.
Please check it 
/////////////////////////////////////////////////////
.
..............................................
:107FA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE1
:107FB000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD1
:107FC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC1
:107FD000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFB1
:107FE000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFA1
:107FF000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF91
:108000000C94C5400C94DB400C94DB400C94DB409A
:108010000C94DB400C94DB400C94DB400C94DB4074
:108020000C94DB400C94DB400C94DB400C94DB4064
:108030000C94DB400C94DB400C94DB400C94DB4054
:108040000C94DB400C94DB400C94DB400C94DB4044
:108050000C94DB400C94DB400C94DB400C94DB4034
:108060000C94DB400C94DB400C94DB400C94DB4024
:108070000C94DB400C94DB400C94DB400C94DB4014
:108080000C94DB400C94DB400C94DB400C94DB4004
:108090000C94DB400C94DB400C94DB400C94DB40F4
:1080A0000C94DB400C94DB400C94DB400C94DB40E4
:1080B0000C94DB400C94DB400C94DB400C94DB40D4
:1080C0000C94DB400C94DB400C94DB400C94DB40C4
:1080D0000C94DB400C94DB400C94DB400C94DB40B4
:1080E0000C94DB400C94DB400C94DB400C94DB40A4
:1080F0000C94DB400C94DB400C94DB400C94DB4094
:108100000C94DB400C94DB400C94DB400C94DB4083
:108110000C94DB400C94DB400C94DB400C94DB4073
:108120000C94DB400C94DB400C94DB400C94DB4063
:108130000C94DB400C94DB400C94DB400C94DB4053
:108140000C94DB400C94DB400C94DB400C94DB4043
:108150000C94DB400C94DB400C94DB400C94DB4033
:108160000C94DB400C94DB400C94DB400C94DB4023
:108170000C94DB400C94DB4058426A01704777470F
:10818000A647D047BD477947E54711241FBECFEF2B
:10819000DFE2DEBFCDBF18BE19BE1ABE1BBE11E2A4
:1081A000A0E0B0E201C01D92A130B107E1F70E944A
:1081B00033440C94EC470C9400403BB78BBFFB015D
:1081C00086913BBF08953BB78BBFFB01879196918A
:1081D0003BBF089542E097C041E095C08093C00145
.
........................................
///////////////////////////////////////////////////////////////////////////////////////////
Jump table? can u explain what exactly the jump table is?

MADHUKARAVS, 2015-09-25:
Also after programming Xboot.hex, i m able to program my application code only once, second tie its giving verification error

MADHUKARAVS, 2015-09-30:
Loaded the xboot.hex file to xmega32a4u and boot file starts from the location 0x8000
 
None of the API are working.
Im totally lost
Added Xbootapi.c and .h  to my application code and called some of the API. None of them working

alexforencich, 2015-10-11:
Are you sure you're compiling both your application and xboot for the atmega32a4u?  There is an important difference between the two wrt. the size of the interrupt vector table.  If this setting disagrees, then the API code will attempt to load the jump table from the wrong offset.

alexforencich, 2015-10-11:
It looks like this is exactly the problem.  iox32a4.h has the following:

    #define _VECTOR_SIZE 4 /* Size of individual vector. */
    #define _VECTORS_SIZE (94 * _VECTOR_SIZE)

while iox32a4u.h has:

    #define _VECTOR_SIZE 4 /* Size of individual vector. */
    #define _VECTORS_SIZE (127 * _VECTOR_SIZE)

Each of the 0C94xxxx in your hex dump corresponds to a jump instruction in the vector table.  I count 94 of them.  The 58426A after the last one is the 'magic number' 'XBj' that the API wrapper in xbootapi.c uses to detect the jump table.  My guess is that your application was compiled for the atmega32a4u and hence has a different value for _VECTORS_SIZE that makes it look in the wrong spot for the table.  Try compiling xboot for the correct device and try again.

MADHUKARAVS, 2015-10-12:
ho, thank you alex for your valueable display.

Yes, just checked the code and as you said, in boot code i selected 32a4 and in my app 32a4u.
i will change the setting  and i will check again.

.
Again i have noticed in boot code If i select    'MCU = atxmega32a4u ' , it will give error as

xboot.c:1: error: MCU 'atxmega32a4u' supported for assembler only

alexforencich, 2015-10-12:
That's very strange.  What version of avr-gcc are you using?

MADHUKARAVS, 2015-10-12:
I am running avr-gcc 4.3.3, avr-binutils 2.19, and avr-libc 1.6.7.

alexforencich, 2015-10-12:
Well, you may need to update that if you're having trouble compiling it.  However, I fail to see why your program would build successfully for the x32a4u when xboot has issues, even though you're using the same make file.  That is very strange.

MADHUKARAVS, 2015-10-19:
thank you, i will update my AVR and will check my code again.
Thank you very much for your valuable reply :)