alexforencich/xboot #27: Make configurations

epccs, 2016-06-06:
I'm probably not understanding how this was setup. I should be able to do 

make arduino328p.conf.mk

to build that target correct? It is not working for me on Linux (Ubuntu 16.04) 

~/Samba/xboot$ make arduino328p.conf.mk
cp arduino328p.conf.mk config.mk
cp: cannot stat 'arduino328p.conf.mk': No such file or directory
Makefile:779: recipe for target 'arduino328p.conf.mk' failed
make: *** [arduino328p.conf.mk] Error 1

I can build when placing those setting directly into the Makefile.

alexforencich, 2016-06-07:
Try

make conf/arduino328p.conf.mk

epccs, 2016-06-07:
That was simple, I was trying stuff like ./conf/arduino328p.conf.mk and /conf/arduino328p.conf.mk

alexforencich, 2016-06-07:
hmm, not sure why ./conf/arduino328p.conf.mk doesn't work.  IIRC the makefile just runs cp to copy the .conf.mk file as conf.mk.

epccs, 2016-06-07:
Now I'm trying to setup a board with a 1284p. So I copy the arduino328p.conf.mk into irrigate7.conf.mk and modify things. It compiles but is not working as expected. Figured the LED would would be a good place to start.

# LED
LED_PORT_NAME         = B
LED_PIN               = 7
LED_INV               = 1

it is at https://github.com/epccs/xboot/blob/master/conf/irrigate7.conf.mk

It does not blink the led... so in xboot.h after #define LED_PORT I added

#ifdef LED_PORT
#define XSTR(x) STR(x)
#define STR(x) #x
#pragma message "The value of LED_PORT: " XSTR(LED_PORT)
#error stop everything while I figure it out
#endif

The idea is to see what the preprocesor is going to feed the compiler...  and it said.

xboot.h:239:9: note: #pragma message: The value of LED_PORT: (*(volatile uint8_t *)((0x05) + 0x20))
 #pragma message "The value of LED_PORT: " XSTR(LED_PORT)
         ^
which looks like PORTB PIN5. 

Any ideas.

epccs, 2016-06-07:
Some more digging and I see what is going on a little better. LED_PORT for mega1284 is made from "PORT" ## 'B' => PORTB => _SFR_IO8(0x05) => (*(volatile uint8_t *)((0x05) + 0x20)). So that was not an issue. I think USE_LED was set wrong. At least it is blinking the expected LED now, but that is as far as I have got.  Avrdude has yet to upload to it. My current config file looks like https://github.com/epccs/xboot/blob/0028098eb3aa648a81925621a0a8b0c7f6f6eb0e/conf/irrigate7.conf.mk

alexforencich, 2016-06-07:
I was going to say, that part looks right to me.  0x25 is the address of PORTB as per the register listing in the datasheet.  

Are you sure that you set the fuses correctly?

epccs, 2016-06-08:
I'm using an Arduino as ISP and programming it in two steps as follows:

avrdude -v -p atmega1284p -c stk500v1 -P /dev/ttyACM0 -b 19200 -e -U lock:w:0x3f:m -U lfuse:w:0xF7:m -U hfuse:w:0xD0:m -U efuse:w:0xFD:m

avrdude -v -p atmega1284p -c stk500v1 -P /dev/ttyACM0 -b 19200 -e -U flash:w:xboot.hex -U lock:w:0x2f:m

perhaps I've got a fuse set wrong...

http://eleccelerator.com/fusecalc/fusecalc.php?chip=atmega1284p&LOW=F7&HIGH=D0&EXTENDED=FD&LOCKBIT=2F

after that, I have a little blink program and have been trying 

avrdude -p atmega1284p -c avr109 -P /dev/ttyUSB0 -b 115200 -e -U flash:w:blink.hex

but no luck so far.

epccs, 2016-06-08:
Excellent, it is mostly working now, I am doing a rather involved full duplex RS485 serial connection and had that a little broken.  Avrdude is not able to read back fuses, I need to go through the config and see if something is turned off. I put the output in a readme if you want to have a look

https://github.com/epccs/irrigate7/tree/df7a2d98f1697a8fa4f0da4198fb912e3a5a5a25/BlinkLED

alexforencich, 2016-06-08:
Well, the bootloader doesn't support accessing the fuse bits, that has to be done when you write the bootloader to the chip in the first place.

epccs, 2016-06-08:
Looks like I can disable "safemode" fuse checking with the avrdude -u option.

avrdude -p atmega1284p -c avr109 -P /dev/ttyUSB0 -b 115200 -e -u -U flash:w:blink.hex

Nice I can now do a makefile without any errors showing.

Thank you for your Time and Efforts on this bootloader, it really helps to have these sort of tools available for my projects.

RLmonitor, 2018-04-16:
I'm also having trouble with the conf make.
I'm on Win10. I moved the Xboot0 folder to D: to avoid a long path. And have tried running make in an Administrator command prompt. Any ideas?

D:\Xboot0>make conf/arduino328p.conf.mk
      0 [main] sh 10024 sync_with_child: child 11456(0x208) died before initialization with status code 0xC0000142
    105 [main] sh 10024 sync_with_child: *** child state waiting for longjmp
/usr/bin/sh: fork: Resource temporarily unavailable
cp conf/arduino328p.conf.mk config.mk
"make"
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 2: syntax error: unexpected end of file
make: *** [conf/arduino328p.conf.mk] Error 258

alexforencich, 2018-04-18:
I have no idea what's breaking there.  Looks like something on your system is unhappy, though I'm not sure what it might be.

RLmonitor, 2018-04-18:
Thanks for your response.
I managed to get it to compile after installing WinAVR and changing a few things, including ...
>creating a dep folder for the dependencies
>editing the make file to change .dep folder references to dep
>manually editing the make file for the atmega328p target

Once I understood everything better, I was also able to compile & link successfully in AS7.

jpentlan, 2018-12-05:
RLmonitor, I am having the exact same issue (word for word error message) as you had in your April 15 post.  I see you solved it in your Apr 17 post.  Could you elaborate on the steps you took to fix the issue.  It would be much appreciated.

RLmonitor, 2018-12-06:
In response to jpentlan ...
It was a while back and I am not currently using Xboot, so I am not sure that I can elaborate further.
I could archive the folders and share them on google drive?