alexforencich/xboot #30: Fails to flash application on xmega384D3

codemonkey1000, 2017-11-05:
I crated a config file for the xmega384D3 and modified the Makefile to have the correct boot loader address. 
the makefile programs the boot loader to the chip and i can enter the boot loader using a button.  When trying to flash the application following your example it fails to program. avrdude says ****failed repeatedly and the verify step at the end fails also.   Any idea why this might happen?

alexforencich, 2017-11-05:
Have you tried with an explicit erase (avrdude -e)?

codemonkey1000, 2017-11-05:
Thanks for the reply. I did not try -e. I  looked at another boot loader,
Xmega boot loader, it didn't work either.
I was worried a bit that i was running into AVR1008 with the D3 chip so i
tried with a C3, same thing.
Ended up being the "page erase" command being sent from avrdude (6.2)
Will try it with xboot later today with the -D and with -e.  I prefer your
make file solution, easier to automate.

Thx

Eric

codemonkey1000, 2017-11-05:
using -D to disable page erase worked.  xboot does flash using xmega384d3.