alexforencich/xboot #35: Problem compiling non midified source in AS 7.1

Blom1974, 2017-11-28:
Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\atmel\xboot-master\xboot-master" -f "Makefile" clean 
		make: Entering directory 'C:**/**atmel**/**xboot-master**/**

xboot-master'
		The system cannot find the path specified.

Noticing the SLASH instead of BACKSLASH in the path but have no clue how to fix it..
So probably not a code problem but thought someone have had this problem before.

alexforencich, 2017-11-29:
Not sure if this is an xboot issue or an AVR studio issue.  Maybe try removing special characters (namely, '-') from the path and see if that has any effect?

Blom1974, 2017-11-29:
Thanks for the fast reply. I am not sure it is a code problem either. Missed to write it in the post though..
Thought maybe someone has had the same problem.

tested to change the xboot-master to just xboot with the same result.

It must have something to do with the slash in the path if you ask me..  Will have to search around a bit.