alexforencich/xboot #36: blocks in erase page

ricardolima91, 2017-12-22:
Hi, Thank you for your bootloader. 
I'am facing a problem using the function xboot_erase_application_page. The code blocks and don't run .. I i'am using in the same code  xboot_app_temp_erase and it is working.. Do we need some extra command do to this ? Thank you 
Thanks in advance

alexforencich, 2017-12-23:
What chip are you using?

ricardolima91, 2017-12-24:
Hi i was using 1284au but i found the problem and probably not related with your code. The code blocks if i call the function install firmware(so the problem was in this function). I created an equal function to this only changing the name and its working ... I dont know exactly where was the error but was not xboot fault. So sorry to bother you and keep the good work