alexforencich/xboot #38: Start bootloader section address

Romarm, 2018-05-16:
I have some misunderstanding with start bootloader section address.
I use atmega644p device. According to "Atmel-8011-8-bit-AVR-Microcontroller-ATmega164P-324P-644P_datasheet" table 23-13 Boot Reset Address is 0x7000 while BOOTSZ1=0 and BOOTSZ0=0.
But in xboot Makefile BOOT_SECTION_start defined as 0xE000.
...
ifneq ($(filter $(MCU_S), m64 m640 m644 m644p m644pa m645 m6450 m649 m649p m6490),)
  ifeq ($(BOOTSZ), 0)
    BOOT_SECTION_START		=0x00E000
    BOOT_SECTION_SIZE		=0x002000
  endif
...
In fact, when I read chip flash after xboot flashing I see code at 0xE000 as defined in xboot Makefile.
What is wrong?

alexforencich, 2018-05-16:
These constants are defined in bytes; the figures in the datasheet are in 16 bit words.  0x7000 in words is 0xE000 in bytes.

Romarm, 2018-05-16:
Oh, Thank You!

closed 2018-05-18 by alexforencich