alexforencich/xboot #42: Generating config.h - syntax error: unexpected end of file

cornedoggen, 2020-10-31:
Hi all, 

I am trying to build xboot with the x32a4u.conf.mk configuration by running:

$ make conf/x32a4u.conf.mk

but i get the following error:

$ make conf/x32a4u.conf.mk
cp conf/x32a4u.conf.mk config.mk
"C:/Program Files (x86)/GnuWin32/bin/make"
make[1]: Entering directory `D:/Programmeren/avr-xboot/xboot'

-------- begin --------
avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1778) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating config.h for atxmega32a4u
/usr/bin/sh: -c: line 1: syntax error: unexpected end of file
make[1]: *** [config.h] Error 1
make[1]: Leaving directory `D:/Programmeren/avr-xboot/xboot'
make: *** [conf/x32a4u.conf.mk] Error 2

It seems something goes wrong in the config.h.mk file ( the makefile part where the config.h is created from the config.mk)
The process of creating config.mk , (just a copy of x32a4u.conf.mk) is working fine. config.mk is there in the folder. 

Any idea?

Kind regards, 
Corne

cornedoggen, 2020-11-02:
@alexforencich please can you take a look into the above issue? I really want to go on with the bootloader and add CAN bus support to it, but first the concept must be working of generating a valid config.h file. Somewhere in the nested calls in config.h.mk there is something going wrong... foreach -> sort -> filter out -> filter... 

Kind regards, 
Corne Doggen

alexforencich, 2020-11-02:
Works fine for me on linux.  Maybe try running under WSL.

cornedoggen, 2020-11-03:
hmm.. i prefer Windows to compile on. I don't really understand the if statement in the last argument of the foreach function. is it a callback argument? I think there is some difference between an bash script (.sh) and a makefile. Can you use all the syntax from bash script in makefile? like if , for, and other? And use them the same ?? I ask this because for an if statement in bash one could use :  

if [  statement ] 

but een if statement in makefile is more seen as:  

ifeq( foo, bar )