alexforencich/xboot #44: Compiling Issues

BuriKizilkaya, 2021-06-25:
Hi

When I compile with the command `make conf/arduino328p.conf.mk` then I receive always the error: 
```
inlining failed in call to always_inline 'uart_deinit': function body not available
 extern void __attribute__ ((always_inline)) uart_deinit(void);
                                             ^
```

What are I do wrong?

alexforencich, 2021-06-25:
Sounds like something may have changed in the compiler and it's not happy with how the code is written.  I ran in to a similar issue not long ago with a project involving RISC-V and had to do some experimenting with various attributes to make it work.  It's possible that xboot has a similar issue with newer versions of gcc.  Not sure I will have time to look at that soon, though.

BuriKizilkaya, 2021-06-25:
I saw in the r2021.03.03 branch do you have this statement in the file uart.h:

```
extern void __attribute__ ((always_inline)) uart_init(void);
extern void __attribute__ ((always_inline)) uart_deinit(void);
```

and in the master branch do you have this statement in the file uart.h:

```
extern void uart_init(void);
extern void uart_deinit(void);
```

alexforencich, 2021-06-25:
Yeah, gcc has evolved quite a bit. And I don't think I set things up correctly in the first place. I think maybe all the inline stuff needs to go away and then LTO needs to be turned on. I will need to experiment with that when I have time.